chris-ali/j6dof-flight-sim #37: Hello

hervegirod, 2017-04-06:
It's not really an issue, I discovered your great project. It works very well but I would like:
- to use the simulation if necessary without the GUI (including the world).  In fact I would like to decouple completely the GUI from the simulation core. Also I would like to be able to use JavaFX in lieu of LWJGL (making the GUI an interface)
- to be able to use the simulation directly just as a library with or without its graphics
- to make the datas directories more configurable
- maybe go from text configuration files  to XML (or even have the two formats possible)

It is a lot of proposed changes(maybe some I propose are really silly too), so I forked your project to be able to play with these ideas without bothering you with unwanted changes on your side. I have a lot of experience in Java / JavaFX development, including JavaFX 3D (not so much with LWJGL / OpenGL I must admit, even if I used JOGL some time ago to make an OpenGL driver for Swing apps). If you are interested, I am willing to merge back the code you will find useful).

Regards,

Hervé Girod

hervegirod, 2017-04-06:
I forgot to say that I dont have a lot of projects on github but I have a lot of them on sourceforge: https://sourceforge.net/u/hervegirod/profile/

hervegirod, 2017-04-06:
A last question, sorry, what license do you use? That way I would use the same one if I add new classes on my fork.

chris-ali, 2017-04-07:
Hey Hervé,

Thank you for your interest in the project, as well as your suggestions. 

That'd be really great if you wanted to split the core simulation into a separate library, since it has plenty of implementation possibilities. When I first started the project, it just consisted of the simulation, without a GUI, graphs or OpenGL world. I added some quick and dirty implementations of these later to try to make it something a pilot could interact with. 

XML configuration would be another nice thing to have. The text files were a quick solution I decided on so that I could quickly import some aerodynamics data from MATLAB. I believe the open source flight dynamics engine JSBSim uses XML to handle their [configuration](http://jsbsim.sourceforge.net/JSBSim/classJSBSim_1_1FGTable.html) as well.

I'm planning on pushing an update to tidy up some of the file paths in the code, integrating Maven, etc. I must admit, I made this project as I was teaching myself Java, so I'm maintaining things here and there as I gain more experience with it.

I definitely look forward to seeing your development, and would be happy to merge some of your changes back into my code later on. If you have questions about anything in the code, don't be afraid to ask.

Also, I checked out some of your projects on Sourceforge. Your ARINC 661 tool is interesting! I take it you do some work with aircraft display units?

chris-ali, 2017-04-07:
As for the license, I'm going with the GNU GPL v3

hervegirod, 2017-04-08:
Hello,

Thanks, you really did a great job!

Yes, I'm working on display Units software for Dassault Aviation. I began to develop the ARINC 661 tool as a personal project some years ago.

I will try to explain what changes I have made in your code in the issues on the forkedprjkect. For the moment:
- I fixed a few errors in javadoc tags which prevented the javadoc generation
- I added a distrib ant target allowing to distribute the bin and source content of the library
- I added a GPL v3 copyright tag with your name as owner for all source files

I hope to begin the decoupling soon ;)

hervegirod, 2017-04-08:
I now have an interface for the world rendering, the LWJGL code implements this interface. I also allowed the decoupling between the class which return the terrain  height and the LWJGL implementation. In fact providing the terrain height is performed through an interface, and the LWJGL implementation happens to implement both this interface and the world rendering class. The simulation still works ;)

hervegirod, 2017-04-09:
The LWJGL world rendering and plotting are now two interfaces, with two libraries which implement them. I also created a default implementation of the code which calculates the height, not depending on the LWJGL library, to be able to compute the simulation without this library. I'm still not sure if I did not forgot something, I will see it soon ;) The simulation as it is (with plotting and LWJGL) continues to work properly.

chris-ali, 2017-04-11:
Sorry for the late reply. I've finally had a chance to see your changes so far, and to try out your latest release. I really like how you've subdivided the project! That's something I'd like to emulate next.

I like the terrain height interface you used when seperating the world renderer. I'll try to integrate that in the future, since I may change my world renderer to something better optimized if I can find one. It's hard to make a good 3D visual engine...

Thanks for adding the license to the source code. When I started, I was lazy about adding a license to my source code, since I didn't imagine that my project would gain any interest, but it appears I was wrong :)

That's interesting that you're at Dassault. I used to work at Gulfstream Aerospace on their G500 and G600 flight simulators, and from time to time I would work on representative display units developed in VAPS XT. A lot of the experience I gained from there inspired this project.

hervegirod, 2017-04-11:
You really did some amazing work! And the performance of your architecture seems to be very good too!

FYI if you look at the test package, I renamed the classes as Test* rather than *Test because I added a JUnit test class to check that I did not forget some dependencies between the packages. And by default JUnit tries to execute all classes ending with Test* in its framework (which did not work with your test classes of course because they were nt designed to work with JUnit).  Also I added 3 "TestStandalone*" classes to test the usage of your library without the menus, or the GUI, for example.

chris-ali, 2017-04-12:
Thank you! I tried as best as I could to optimize the simulation calculations, as you would need a really high recalculation rate (100-200 Hz) when working with things like control laws and autopilots.

On a powerful computer like my desktop it the visuals run fine, but the frame rate really drops on my older laptop. I've been trying to come up with some way of selectively rendering objects that are in range of the aircraft. It works for a while, but video memory usage still gradually creeps up. It's definitely a work in progress, but also an interesting learning experience!

Oh yeah, I forgot about my naming convention with those tests. Back when I made them I didn't know that JUnit could use class name to find classes to execute. Thanks for changing their names.

treemark, 2018-03-06:
Hello Chris, I like what you've laid down here. I'm working on a related project involving UAVs, and see an opportunity to collaborate as I have built a global 3d terrain model, but need a good flight model to simulate common hobby aircraft. If you're interested, email me at treemark@gmail.com and I'll share some videos of what I have.