chris-ali/j6dof-flight-sim #58: Documentation.docx does not match unzipped folder contents.

Iknowlongnamerightwhatwasithinking, 2019-01-11:
There is no "RunJavaFlightSimulator.bat"  so there is mis match between "documentaion.docx" and actual files in folder. 

So yeah.., like how do we start the program? 
Documentation, although is pretty good, fails to address this. 

Press "E" to start? ( Flight sim joke ) 

Would love to try this out but no idea how to start it. 
Documentation needs updating. 

Have Java installed. No a Java user. Just would like to try the sim. 

Thanks! 

R.A.

chris-ali, 2019-01-12:
Hey R.A.!

Sorry for the delay, just got a chance to double check and make sure I didn't forget that file in the release. RunJavaFlightSimulator.bat is indeed included in the file JFS-0.4a.zip.

I think Documentation.docx is also in the source code, so if you downloaded the source code zip file, it would seem like that was the actual built release.

Once you run the .bat file, it should be pretty straight forward to start the simulation, like pressing E (or was it CTRL-E? It's been a while since I used FSX) to start engines!

Chris

Iknowlongnamerightwhatwasithinking, 2019-01-12:
Oh ... it's in the "release" tab not the "clone or download" button. Jeepers. 

The DL button is misleading! So I DL'd the uncompiled "source code" ... or something. 

OK. I get it now. Sheesh. They should make a better "download latest release" button. and make it HUGE. 

Thanks!!!! 

Will try out your sim and get back to you. Tell me it supports head tracking... please.... 

Robert Alexander.  : -)

chris-ali, 2019-01-12:
Robert,

No problem! I found it pretty tough when I was first starting out on GitHub to do things like check in code changes or download releases, but once you learn the GItHub way, things start to make a lot more sense.

Unfortunately I don't have head tracking support at the moment. I've been meaning to get a 3D interior view and camera implemented, but the math to get everything looking right in that perspective is pretty  tricky.

Chris

closed 2019-01-12 by chris-ali

Iknowlongnamerightwhatwasithinking, 2019-01-12:
Hi Chris. 

Yeah I know what you mean.( 3d in cockpit view)  I am a Rise of Flight fan big time and 
I really hate how there is no "foreshortening" in first person view AND 
the "everthing is bigger at the edges"  distortion. No one seems to care!!!! oh well. 

Hey Can't get my Logitech extreme 3d ( highly customised...) stick to work. 
How do I setup the config file for it??? I tried but had mixed results. 

This is my baby. It's amazing. : - ) 

[image]

I managed to get pitch and roll but no rudder and then I broke the .json file... 

Thanks! 

Robert Alexander.

chris-ali, 2019-01-12:
Robert,

That's a really cool setup you have there, good job with the custom stick and rudder! I imagine it nails the feel of an early WW1 fighter pretty well (makes sense, given that you are a fan of Rise of Flight).

Personally I was never a fan of combat flight sims. I always preferred peaceful civil flight sims, but they could get pretty stressful as well when you're hand-flying an approach in bad weather and dealing with equipment failures.

As for the config issue, JSON can be kind of a pain when it comes to adding in extra rows, since it's very particular when it comes to syntax. When adding a new line for the rudder, did you miss a comma by chance? The log file in ./logs/javaflightsim.log would say something like:

`[ERROR] 2019-01-12 08:54:15.652 [AWT-EventQueue-0] FileUtilities - Could not read: ControlsConfiguration.json!
com.fasterxml.jackson.databind.JsonMappingException: Unexpected character ('"' (code 34)): was expecting comma to separate Object entries`

This:
"axisAssignments" : {
        "y" : {
          "axisAssignment" : "ELEVATOR",
          "deadZone" : 0.0,
          "sensitivity" : 1.0
        }
        "z" : {
          "axisAssignment" : "RUDDER",
          "deadZone" : 0.0,
          "sensitivity" : 1.0
        }
      },
would cause a parsing error because of the missing comma between `}` and `"z"`. But the last item in a list doesn't have a comma at the end, which can get confusing when copying and pasting.

A GUI for setting up keys, buttons and axes is definitely in the works for the next version!

Chris

Iknowlongnamerightwhatwasithinking, 2019-01-12:
{
  "keyboardAssignments" : {
    "DOWN" : "ELEVATOR_UP",
    "LEFT" : "AILERON_LEFT",
    "G" : "GEAR_UP_DOWN",
    "L" : "GENERATE_PLOTS",
    "PG UP" : "INCREASE_THROTTLE",
    "P" : "PAUSE_UNPAUSE_SIM",
    "F6" : "DECREASE_FLAPS",
    "Q" : "EXIT_SIMULATION",
    "F7" : "INCREASE_FLAPS",
    "R" : "RESET_SIM",
    "PG DOWN" : "DECREASE_THROTTLE",
    "RIGHT" : "AILERON_RIGHT",
    "UP" : "ELEVATOR_DOWN"
  },
  "joystickAssignments" : {
	"Logitech Extreme 3D" : {
	  "axisAssignments" : {
        "y" : {
          "axisAssignment" : "ELEVATOR",
          "deadZone" : 0.0,
          "sensitivity" : 1.0
        },
        "x" : {
          "axisAssignment" : "AILERON",
          "deadZone" : 0.0,
          "sensitivity" : 1.0
        }
      },
      "buttonAssignments" : {
        "0" : "ELEVATOR_TRIM_DOWN",
        "1" : "ELEVATOR_TRIM_UP"
      },
      "hatAssignments" : {
        "0.25" : "ELEVATOR_TRIM_DOWN",
        "0.75" : "ELEVATOR_TRIM_UP",
        "1.0" : "AILERON_TRIM_LEFT",
        "0.5" : "AILERON_TRIM_RIGHT"
      }
	}
  }
}

Will try to get the rudder workign next.  
The d in "3d" needed to be upper case... "3D" this was the issue!!! 

R.A.

chris-ali, 2019-01-13:
Robert,

You're quite  welcome! 

I'd imagine that RoF would need a really good flight model since it's all stick and rudder in those early aircraft. A lot of aerobatic maneuvers in dogfights happen at the extreme ends of an aircraft's aerodynamic capabilities, so they probably took a lot of time to model things like stalls and spins. I tried my best in my sim, but accurate data used in lookup tables for the aerodynamics is really tough to come by.

Seems like you can get some fine control adjustments from having a long arm for the joystick. Ordinary joysticks can be pretty twitchy until you turn the sensitivity down.

Good to hear that you got the joystick recognized in the sim! As for the rudder assignment, try this out (if you haven't already):

{
"keyboardAssignments" : {
"DOWN" : "ELEVATOR_UP",
"LEFT" : "AILERON_LEFT",
"G" : "GEAR_UP_DOWN",
"L" : "GENERATE_PLOTS",
"PG UP" : "INCREASE_THROTTLE",
"P" : "PAUSE_UNPAUSE_SIM",
"F6" : "DECREASE_FLAPS",
"Q" : "EXIT_SIMULATION",
"F7" : "INCREASE_FLAPS",
"R" : "RESET_SIM",
"PG DOWN" : "DECREASE_THROTTLE",
"RIGHT" : "AILERON_RIGHT",
"UP" : "ELEVATOR_DOWN"
},
"joystickAssignments" : {
"Logitech Extreme 3D" : {
"axisAssignments" : {
"y" : {
"axisAssignment" : "ELEVATOR",
"deadZone" : 0.0,
"sensitivity" : 1.0
},
"x" : {
"axisAssignment" : "AILERON",
"deadZone" : 0.0,
"sensitivity" : 1.0
},
"z" : {
"axisAssignment" : "RUDDER",
"deadZone" : 0.0,
"sensitivity" : 1.0
}
},
"buttonAssignments" : {
"0" : "ELEVATOR_TRIM_DOWN",
"1" : "ELEVATOR_TRIM_UP"
},
"hatAssignments" : {
"0.25" : "ELEVATOR_TRIM_DOWN",
"0.75" : "ELEVATOR_TRIM_UP",
"1.0" : "AILERON_TRIM_LEFT",
"0.5" : "AILERON_TRIM_RIGHT"
}
}
}
}

Usually joysticks that rotate left and right use the z axis for that, though you can verify which one exactly the Logitech uses in the game controller calibration menu in the Control Panel (I believe you can get to it from Devices and Printers as well).

Chris

Iknowlongnamerightwhatwasithinking, 2019-01-13:
Hi! , please I need to invert Pitch input... I can do it with stick to joystick Gremlin to Vjoy but I would rather not. 
and I am picking my rudder will need to be inverted too.

Hey have you read the Wiki on "Flight unlimited" and Seamus Blackly? What a god. 
https://en.wikipedia.org/wiki/Flight_Unlimited

Robert Alexander.